UTNuclearRoboticsPublic/jog_arm #1: Question about param  pub_period ?

l0or, 2017-12-26:
I noticed there are two thread in _jog_arm_server.cpp_ code. In the main thread the loop using spinOnce() to  deal with the callbacks of topics about joint_state and joy_vel. I don't know If it means to update the share variables  every 0.01 seconds(default value of pub_period)?But as far as i know, the publish rate of joint_states may up to 150Hz. Does this mean some joint_state message will be overflowed and ignored?

AndyZe, 2017-12-26:
Yep, if that parameter is set to 0.01, some publications at 150 Hz would be ignored. Change it to  0.006 to be sure to catch every new message. There won't be any overflow issues, though.

l0or, 2017-12-26:
@AndyZe  Thanks for the fast reply ! Really apreciate !!

closed 2017-12-26 by AndyZe