UTNuclearRoboticsPublic/jog_arm #11: Better position mode support for closed loop systems

machinekoder, 2018-03-22:
As of now `jog_arm` resets the internal joint state when a new `joint_status` message is received. This causes problems when the movement is still active as suddenly the joint trajectory position is reset to the current joint state -> this results in a jump in the movement.

To properly support position mode commands, `jog_arm` would need to keep track of the joint state internally and reset it for example after a certain timeout,

I don't think this problem happens when the update interval is very small, as there will not be any feedback messages before the next command is executed. However, my controller just ignores commands with outdated timestamps, so having a very short update interval doesn't work out as there is still a network delay of a few ms between the sender (joystick) and the receiver (controller).

machinekoder, 2018-03-22:
Another option would be to not increase the internal joint position and just relying on `joint_state` updates. At least this would prevent backward jumps form happening.

AndyZe, 2018-03-22:
OK, I'll work on this. Can I ask what type of arm you're using?

machinekoder, 2018-03-22:
Custom 6 axis with `roscontrol` + Machinekit. It's in very early stage right now.

machinekoder, 2018-03-22:
Thanks for taking a look at this. Let me know how I can help.

AndyZe, 2018-03-23:
Let's see if I understand the problem... the internal` jt_state_ `gets updated more frequently than new `/joint_states` msgs come in. So it is cranked up fairly high before being reset by a new `/joint_states` msg.

On my setup, jt_state_ is updated 10X before being updated with a new `/joint_states` msg.

I propose to solve this by low-pass filtering the `jt_state_` positions as well as the velocities.

AndyZe, 2018-03-23:
Please give 19abbbc a try. I think you will still see a small jump backwards when a new `/joint_states` msg arrives but hopefully the magnitude is much smaller.

You could also increase the value of `low_pass_filter_coeff` in the launch file. Maybe try ~10. That will smooth the trajectory more.

AndyZe, 2018-03-23:
Actually, for me, this eliminated the backwards jumps.  :)

machinekoder, 2018-03-26:
I tested again and found out that backward jumps happen with a very big `pub_period` (0.2s). After reducing the time `pub_period` to 0.01 the problem is gone.