UTNuclearRoboticsPublic/jog_arm #45: Share efforts with jog_control

machinekoder, 2018-07-02:
There is a new package called [jog_control](https://github.com/tork-a/jog_control). Looks like there are double efforts going on in the Cartesian jogging space at the moment. It might be interesting to share efforts.

AndyZe, 2018-07-05:
I've had the same thought, but jog_arm is close to being merged into [MoveIt!](https://github.com/ros-planning/moveit/pull/854) So I think I will wait for that. There was already a simple [XBox jogger](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/joystick_control_teleoperation/joystick_control_teleoperation_tutorial.html) in MoveIt!, too ... whereas this package is for high-frequency, real-time control.

I'll give it a couple weeks to see how the MoveIt situation works out.

AndyZe, 2018-07-16:
We are getting close to the time when I will give up on the MoveIt! PR. They are moving too slowly and it's hard to maintain this repo + that PR. Will be thinking more about this...

gavanderhoorn, 2019-02-08:
A big difference between this package and `jog_control` is that the latter relies on the presence of a `FollowJointTrajectory` action server, and I believe even one which supports trajectory replacement for jogging to work smoothly. In simulation, that all works fine with the `joint_trajectory_controller` loaded, but I'm not sure how well it works with real robots and drivers that don't support trajectory replacement.