UTNuclearRoboticsPublic/jog_arm #60: control a UR5 robot arm with Leap Motion or Kinect non-contact sensors

stevensu1838, 2018-08-22:
Hi sir,
I am planning to control a UR5 robot arm with Leap Motion or Kinect non-contact sensors in real time in ROS. I am an absolute beginner. May I please ask if I can use your package to developt it? Now, I am able be control the UR5 arm very well in Rviz and Gazebo by doing motion planning with Move it. (by following this tutorial http://wiki.ros.org/ur_gazebo)However, what I wanna do is real time teleoperation not first plan the trajectory then execute. Can you please tell me how I can do real time control of UR5 arm in ROS?
secondly, I've noticed in Rivz you can simply drag the tool centre point of UR5 to do the motion planning before execute. Can I only control the tool centre point in the teleoperation to control the whole arm? Coz I only care about the trajectory of the tool centre point. Or I am going have to do mapping for all the joints of the robot to control in real time?
Thanks a lot.﻿

AndyZe, 2018-08-26:
Hi Steven,
Sorry about being a bit slow to reply. Feel free to use this package. Getting MoveIt! to work is half the battle, so good job.

To control the robot in real time, launch the robot in Gazebo and MoveIt! like usual. Then edit this config file: _jog_arm/config/jog_settings.yaml_. The settings should be similar to the attached text file:

[ur5_gazebo_settings.txt](https://github.com/UTNuclearRoboticsPublic/jog_arm/files/2321982/ur5_gazebo_settings.txt)

If it doesn't work right away, make sure the topic names are correct. Try _rostopic list_ or _rqt_graph_ commands to see the ROS topics.

Can you buy an XBox controller to test with? LeapMotion hasn't been tried yet. Then just do _roslaunch jog_arm jog_with_xbox.launch_.

No problem, right?

Once everything is working OK, I would recommend copying and modifying jog_arm/src/jog_arm/teleop_examples/xbox_to_twist to work with the LeapMotion controller.

AndyZe, 2018-08-26:
And you only need to worry about controlling the tool center point. The package does the math to calculate the rest of the joints for you. The math is actually not too complicated (Jacobian inversion).

AndyZe, 2018-09-13:
I'm going to assume this was resolved. Please re-open it, if not.

closed 2018-09-13 by AndyZe