UTNuclearRoboticsPublic/jog_arm #61: Could not find a package configuration file provided by "Eigen3"

stevensu1838, 2018-08-23:
Hi team,
Ubuntu 14.04 ROS indigo
I am experience the following error after install, can you please help me out??

CMake Error at jog_arm/jog_api/CMakeLists.txt:18 (find_package):
  By not providing "FindEigen3.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Eigen3", but
  CMake did not find one.

  Could not find a package configuration file provided by "Eigen3" with any
  of the following names:

    Eigen3Config.cmake
    eigen3-config.cmake

  Add the installation prefix of "Eigen3" to CMAKE_PREFIX_PATH or set
  "Eigen3_DIR" to a directory containing one of the above files.  If "Eigen3"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/ysu66/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/ysu66/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
ysu66@mech1331:~/catkin_ws$

stevensu1838, 2018-08-23:
[image]

AndyZe, 2018-08-26:
Can you please try this and report back if it works:
_sudo apt install libeigen3-dev_

I spend almost all of my time on ROS Kinetic these days, so I can't debug this directly.

AndyZe, 2018-08-27:
There's a good chance this commit (4dd9af0) fixed the issue. Another user encountered it today.

AndyZe, 2018-09-13:
I'm going to assume this was resolved. Please re-open it, if not.

closed 2018-09-13 by AndyZe