UTNuclearRoboticsPublic/jog_arm #75: Cannot roslaunch jog_arm successfully. Please help me out

stevensu1838, 2018-11-07:
Hi Sir, 
PC config: ROS Kinetic on Ubuntu 16.04
Pkg Branches: iron-ox fork of ur_modern_driver (Kinetic-devel branch)
                         jog_arm(Kinetic branch)
First I run in Simulation with the following lines:
- roslaunch ur_gazebo ur5.launch limited:=true
- roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch sim:=true limited:=true
- roslaunch ur5_moveit_config moveit_rviz.launch config:=true    

Next I modified [your attached jog_settings.yaml ](https://github.com/ros-industrial/ur_modern_driver/issues/219) in gazebo and command_out_topic part and my yaml file looks like this:
```
**gazebo: true # Originally, it was false in your attached file** 
collision_check: true 
command_in_type: "unitless" 
scale: 
  linear:  0.002  
  rotational:  0.005  
  joint: 0.006  
cartesian_command_in_topic:  jog_arm_server/delta_jog_cmds 
joint_command_in_topic: jog_arm_server/joint_delta_jog_cmds 
command_frame:  ee_link  
incoming_command_timeout:  5  
joint_topic:  joint_states
move_group_name:  manipulator  
lower_singularity_threshold:  30  
hard_stop_singularity_threshold: 45 
lower_collision_proximity_threshold: 0.1 
hard_stop_collision_proximity_threshold: 0.005 
planning_frame: base_link  
low_pass_filter_coeff: 6.  
publish_period: 0.005  
publish_delay: 0.001  
warning_topic: jog_arm_server/warning
joint_limit_margin: -0.1 
**command_out_topic:  arm_controller/command # Originally, it was ur_driver/joint_speed**
command_out_type: trajectory_msgs/JointTrajectory
publish_joint_positions: false
publish_joint_velocities: true
publish_joint_accelerations: false
```

btw, I've got a question about setting this yaml file. Can you please show me what the difference is of setting it for simulation and real robot respectively?  Such as command_out_topic, I've seen settings like arm_controller/command; ur_driver/joint_speed; sia5_controller/command. This has been confusing.

Afterwards, I put  [ your attached launch and config](https://github.com/ros-industrial/ur_modern_driver/issues/219) in directory /home/user/catkin_ws/Jog_arm_config and I changed  the launch file and made it pointing toward the right yaml file:
`rosparam command="load" file="$(find Jog_arm_config )/jog_settings.yaml" (I tried file="$/home/user/catkin_ws/Jog_arm_config/jog_settings.yaml" as well and it didn't work neither. I might made a silly mistake here , coz I don't exactly know how to point a file when it is not actually put in a ROS package. Can you help me out?)`

Finally, tried to roslaunch jog_node.launch with the following: 
```
user@mech1349:~/catkin_ws$ source devel/setup.bash
user@mech1349:~/catkin_ws$ cd Jog_arm_config/
user@mech1349:~/catkin_ws/Jog_arm_config$ roslaunch jog_node.launch
```
It turns out ......
```
ResourceNotFound: catkin_ws
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/user/catkin_ws/src
ROS path [2]=/opt/ros/kinetic/share

```
So I move on to another way to launch it I use roslaunch jog_arm jog_with_xbox.launch to launch it directly form you Jog_arm. But it throw errors like this:

```
NODES
  /
    jog_arm_server (jog_arm/jog_arm_server)
    joy_node (joy/joy_node)
    xbox_to_twist (jog_arm/xbox_to_twist)

ROS_MASTER_URI=http://localhost:11311

process[joy_node-1]: started with pid [21556]
process[xbox_to_twist-2]: started with pid [21557]
process[jog_arm_server-3]: started with pid [21571]
[ERROR] [1541577097.248599003]: Couldn't open joystick /dev/input/js0. Will retry every second.
[ INFO] [1541577097.323917235]: Loading robot model 'ur5'...
[ WARN] [1541577097.323992265]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
[ INFO] [1541577097.324015867]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1541577097.898556759, 3096.691000000]: Loading robot model 'ur5'...
[ WARN] [1541577097.898704453, 3096.692000000]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
[ INFO] [1541577097.898722744, 3096.692000000]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1541577097.926738277, 3096.717000000]: Loading robot model 'ur5'...
[ WARN] [1541577097.926801273, 3096.717000000]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
[ INFO] [1541577097.926857407, 3096.717000000]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1541577097.931447767, 3096.724000000]: Waiting for first joint msg.
[ INFO] [1541577097.960109311, 3096.753000000]: Loading robot model 'ur5'...
[ WARN] [1541577097.960146028, 3096.753000000]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
[ INFO] [1541577097.960182352, 3096.753000000]: No root/virtual joint specified in SRDF. Assuming fixed joint
[FATAL] [1541577097.970789099, 3096.763000000]: Group 'arm' was not found.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Group 'arm' was not found.
[jog_arm_server-3] process has died [pid 21571, exit code -6, cmd /home/user/catkin_ws/devel/lib/jog_arm/jog_arm_server __name:=jog_arm_server __log:=/home/user/.ros/log/ba9d8b54-e25a-11e8-af8b-4c72b9268f33/jog_arm_server-3.log].
log file: /home/user/.ros/log/ba9d8b54-e25a-11e8-af8b-4c72b9268f33/jog_arm_server-3*.log
^C[joy_node-1] killing on exit
[xbox_to_twist-2] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```
Can you please kindly walk me through how to roslaunch your package successfully?
Sorry about the lengthy explanation  and please let me summarize a bit:
Q 1 how do I modify the setting yaml.file for a UR5 arm differently for both simulation and the real robot? Much appreciated, if you could attach your jog_node and yaml files of UR5!
Q 2 A dummy question. How do point a yaml in my launch file when it is not put in a package in catkin_ws/ src? 
Q3 How shall I exactly launch the pkg? by roslaunch jog_arm jog_with_xbox.launch or  roslaunch jog_node.launch? My target for now it to jog the UR5 with a xbox/3D mouse and then I would move on to Leap Motion.

Thank you so much for your time and kindness. I like your package very much and I hope to make use of it. Thank you again.

Steven
New Zealand

stevensu1838, 2018-11-07:
Hi Andy
Thanks a lot for your prompt reply. I still get stuck at pointing the yaml file.
First, let me show you again my yaml file directory:
[image]
Actually, I followed your instruction to modified my jog_node.launch.file as below:
<launch>

```
  <node name="jog_arm_server" pkg="jog_arm" type="jog_arm_server" output="screen" >
    <param name="parameter_ns" type="string" value="jog_arm_server" />
    <rosparam command="load" file="/home/$USER/catkin_ws/Jog_arm_config/jog_settings.yaml" />
  </node>

</launch>
```

However, when I roslaunch it, I got the following errors:
```
user@mech1349:~/catkin_ws$ source devel/setup.bash
user@mech1349:~/catkin_ws$ cd Jog_arm_config/
user@mech1349:~/catkin_ws/Jog_arm_config$ roslaunch jog_node.launch
... logging to /home/user/.ros/log/fd3721b6-e2e4-11e8-af8b-4c72b9268f33/roslaunch-mech1349-28359.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

error loading <rosparam> tag: 
	file does not exist [/home/$user/catkin_ws/Jog_arm_config/jog_settings.yaml]
XML is <rosparam command="load" file="/home/$user/catkin_ws/Jog_arm_config/jog_settings.yaml"/>
The traceback for the exception was written to the log file
```
```
user@mech1349:~/catkin_ws/Jog_arm_config$ cd ..
user@mech1349:~/catkin_ws$ source devel/setup.bash
user@mech1349:~/catkin_ws$ cd Jog_arm_config/
user@mech1349:~/catkin_ws/Jog_arm_config$ roslaunch jog_node.launch
... logging to /home/user/.ros/log/fd3721b6-e2e4-11e8-af8b-4c72b9268f33/roslaunch-mech1349-28432.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

error loading <rosparam> tag: 
	file does not exist [/home/$USER/catkin_ws/Jog_arm_config/jog_settings.yaml]
XML is <rosparam command="load" file="/home/$USER/catkin_ws/Jog_arm_config/jog_settings.yaml"/>
The traceback for the exception was written to the log file
```
user@mech1349:~/catkin_ws/Jog_arm_config$ 

Can you please help me out? Cheers

Steven

stevensu1838, 2018-11-08:
Hi Andy,

I am getting there and now I am configuring a Xbox360 controller to test it by following Joy node [wiki](http://wiki.ros.org/joy/Tutorials/ConfiguringALinuxJoystick) And everything went well.
However I got errors when I run  
```
$ rosrun joy joy_node
[ INFO] [1541653009.395673744]: Opened joystick: /dev/input/js0. deadzone_: 0.050000.
Segmentation fault (core dumped)
```
It is not related to Jog_arm but It takes me the whole day to proceed. Have you got this issue before??
Cheers,

Steven

AndyZe, 2018-11-08:
Maybe you need to do **sudo apt install ros-kinetic-rosconsole-bridge**?

stevensu1838, 2018-11-09:
Hi Andy, 
You advice on sudo apt install ros-kinetic-desktop-full helped me fix the problem on my PC.
The joy_node works well except [ERROR] [1541720177.655406349]: Couldn't open joystick force feedback!
```
When I run ~$ rostopic echo /joy
I do get he following:
header: 
  seq: 2811
  stamp: 
    secs: 1511
    nsecs: 310000000
  frame_id: ''
axes: [-0.0, -0.0, 0.0, -0.0, 0.005064466502517462, 0.0, -0.0, -0.0]
buttons: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
```

So I ignored [ERROR] [1541720177.655406349]: Couldn't open joystick force feedback! and moved on.

    ```
    roslaunch ur_gazebo ur5.launch limited:=true
    roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch sim:=true limited:=true
    roslaunch ur5_moveit_config moveit_rviz.launch config:=true

```
   ` roslaunch jog_arm jog_with_xbox.launch (I've modified the launch file to point to /home/$USER/catkin_ws/Jog_arm_config/jog_settings.yaml )
`
   ```
 roslaunch jog_arm jog_with_xbox.launch
... logging to /home/user/.ros/log/72cc5cb4-e3b4-11e8-92f7-4c72b9268f33/roslaunch-mech1349-25655.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:43757/

SUMMARY
========

PARAMETERS
 * /jog_arm_server/cartesian_command_in_topic: jog_arm_server/de...
 * /jog_arm_server/collision_check: True
 * /jog_arm_server/command_frame: ee_link
 * /jog_arm_server/command_in_type: unitless
 * /jog_arm_server/command_out_topic: arm_controller/co...
 * /jog_arm_server/command_out_type: trajectory_msgs/J...
 * /jog_arm_server/gazebo: True
 * /jog_arm_server/hard_stop_collision_proximity_threshold: 0.005
 * /jog_arm_server/hard_stop_singularity_threshold: 45
 * /jog_arm_server/incoming_command_timeout: 5
 * /jog_arm_server/joint_command_in_topic: jog_arm_server/jo...
 * /jog_arm_server/joint_limit_margin: -0.1
 * /jog_arm_server/joint_topic: joint_states
 * /jog_arm_server/low_pass_filter_coeff: 6.0
 * /jog_arm_server/lower_collision_proximity_threshold: 0.1
 * /jog_arm_server/lower_singularity_threshold: 30
 * /jog_arm_server/move_group_name: manipulator
 * /jog_arm_server/parameter_ns: jog_arm_server
 * /jog_arm_server/planning_frame: base_link
 * /jog_arm_server/publish_delay: 0.001
 * /jog_arm_server/publish_joint_accelerations: False
 * /jog_arm_server/publish_joint_positions: True
 * /jog_arm_server/publish_joint_velocities: True
 * /jog_arm_server/publish_period: 0.005
 * /jog_arm_server/scale/joint: 0.006
 * /jog_arm_server/scale/linear: 0.002
 * /jog_arm_server/scale/rotational: 0.005
 * /jog_arm_server/warning_topic: jog_arm_server/wa...
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

[... 111 lines cut ...]
[ INFO] [1541719399.711035453, 287.888000000]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1541719399.864272936, 288.041000000]: Received first joint msg.
[ INFO] [1541719399.864314391, 288.041000000]: Waiting for first command msg.
[ WARN] [1541719400.568619221, 288.745000000]: 
Deprecation warning: Trajectory execution service is deprecated (was replaced by an action).
Replace 'MoveGroupExecuteService' with 'MoveGroupExecuteTrajectoryAction' in move_group.launch
[ INFO] [1541719400.571282245, 288.748000000]: Ready to take commands for planning group manipulator.
[ INFO] [1541719400.576106268, 288.752000000]: Waiting for first joint msg.
[ INFO] [1541719400.584702829, 288.761000000]: Received first joint msg.
[ INFO] [1541719400.584740269, 288.761000000]: Waiting for first command msg.
[ INFO] [1541719401.527577170, 289.702000000]: Received first command msg.
[ INFO] [1541719401.527672627, 289.702000000]: Received first command msg.
[ WARN] [1541719401.579174894, 289.753000000]: Stale joint trajectory msg. Try a larger 'incoming_command_timeout' parameter.
[ WARN] [1541719401.579215201, 289.753000000]: Did input from the controller get interrupted? Are calculations taking too long?
^C[jog_arm_server-1] killing on exit
terminate called after throwing an instance of 'class_loader::LibraryUnloadException'
  what():  Attempt to unload library that class_loader is unaware of.
shutting down processing monitor...
... shutting down processing monitor complete
done

```
Can you please walk me through the last obstacle to jog the arm as smoothly as you did in the demo?
![image](https://user-images.githubusercontent.com/25648734/48235737-edd5cc00-e423-11e8-9231-65f1600e5f0a.png)
Thanks a lot!

stevensu1838, 2018-11-09:
Hi Andy,
I've got to apologize for having you read the long comments. I just wanted to offer you as much info as possible to save your time. In case the problem is from my yalm file, please take a look at my xbox.launch file and jog_setting file:
<launch>

```
  <node name="joy_node" pkg="joy" type="joy_node" />

  <node name="xbox_to_twist" pkg="jog_arm" type="xbox_to_twist" output="screen" />

  <node name="jog_arm_server" pkg="jog_arm" type="jog_arm_server" output="screen" >
    <param name="parameter_ns" type="string" value="jog_arm_server" />
    <rosparam command="load" file="/home/user/catkin_ws/Jog_arm_config/jog_settings.yaml" />
  </node>

</launch>
```

```
gazebo: true # Whether the robot is started in a Gazebo simulation environment
collision_check: true # Check collisions?
command_in_type: "unitless" # "unitless"> in the range [-1:1], as if from joystick. "speed_units"> cmds are in m/s and rad/s
scale: # Only used if command_in_type=="unitless"
  linear:  0.002  # Max linear velocity. Meters per publish_period. Units is [m/s]
  rotational:  0.005  # Max angular velocity. Rads per publish_period. Units is [rad/s]
  joint: 0.006  # Max joint angular/linear velocity. Rads or Meters per publish period. Units is [rad/s] or [m/s].
cartesian_command_in_topic:  jog_arm_server/delta_jog_cmds # Topic for xyz commands
joint_command_in_topic: jog_arm_server/joint_delta_jog_cmds # Topic for angle commands
command_frame:  ee_link  # TF frame that incoming cmds are given in
incoming_command_timeout:  5  # Stop jogging if X seconds elapse without a new cmd
joint_topic:  joint_states
move_group_name:  manipulator  # Often 'manipulator' or 'arm'
lower_singularity_threshold:  30  # Start decelerating when the condition number hits this (close to singularity)
hard_stop_singularity_threshold: 45 # Stop when the condition number hits this
lower_collision_proximity_threshold: 0.1 # Start decelerating when a collision is this far [m]
hard_stop_collision_proximity_threshold: 0.005 # Stop when a collision is this far [m]
planning_frame: base_link  # The MoveIt! planning frame. Often 'base_link'
low_pass_filter_coeff: 6.  # Larger --> trust the filtered data more, trust the measurements less.
publish_period: 0.005  # 1/Nominal publish rate [seconds]
publish_delay: 0.001  # delay between calculation and execution start of command
# Publish boolean warnings to this topic
warning_topic: jog_arm_server/warning
joint_limit_margin: -0.1 # added as a buffer to joint limits [radians]. If moving quickly, make this larger.
command_out_topic: arm_controller/command
# What type of topic does your robot driver expect?
# Currently supported are std_msgs/Float64MultiArray (for ros_control JointGroupVelocityController)
# or trajectory_msgs/JointTrajectory (for Universal Robots and other non-ros_control robots)
command_out_type: trajectory_msgs/JointTrajectory
# Can save some bandwidth as most robots only require positions or velocities
publish_joint_positions: true
publish_joint_velocities: true
publish_joint_accelerations: false

```
For UR gazebo simulation, is my yaml file all right? 
command_out_topic is set to  arm_controller/command 
command_out_type is set to trajectory_msgs/JointTrajectory

How shall I set command_frame:  ee_link or base link?
Andy, Thank you so much for walking me through you package. Will never forget your kindness and constant support. Cheers

Steven

AndyZe, 2018-11-09:
Hi Steven, if the arm moved, the topics are correct, so don't worry about that. Either ee_link or base_link is fine when you're just testing if it works. Later you might want to think about whether to control the end-effector frame or from a global frame like base_link.

You can ignore this warning. It will always occur with Gazebo:
`[ WARN] [1541723070.671399927, 43.873000000]: Dropping first 6 trajectory point(s) out of 30, as they occur before the current time.
First valid point will be reached in 0.002s.`

This warning will occur when you stop pressing the joystick, so it's fine:
`Stale joint trajectory msg. Try a larger 'incoming_command_timeout' parameter.`

So, all of your terminal output looks OK.

The UR Gazebo controllers can be a little flaky. I would recommend trying it on a real robot at this point (while staying away from the robot, using small scales in the yaml file, and keeping the E-stop close by). When I have more time, I can send you a tuned UR5 Gazebo simulation. I have done this in the past but don't have it handy at the moment.

stevensu1838, 2018-11-11:
Hi Andy,
Thanks a lot for your great support. Would move on to the real robot ASAP(within 2 days). Meanwhile, I would reference your xbox_to_twist code to write my own leapmotion_to_twist.cpp. My target is to use the fingertip position/hand position to jog the TCP of UR5.
When it come to xbox_to_twist.cpp code, may I please ask why geometry_msg/TwistStamped type is used instead of geometry_msg/Twist?
Also, Would you mind explaining a bit about this part under Cartesian jogging in your code? Coz I could not find info about msg->buttona/msg->axes online and I think that's the part of the code where I need to feed in data from leap motion. Thanks a lot. I am new to C++ but I am trying hard to sort it out. 
 // This button is binary
    twist.twist.linear.x = -msg->buttons[4] + msg->buttons[5];
    // Double buttons
    twist.twist.linear.y = msg->axes[0];
    twist.twist.linear.z = msg->axes[1];
    twist.twist.angular.x = -msg->axes[3];
    twist.twist.angular.y = msg->axes[4];
    // A binary button
    twist.twist.angular.z = -msg->buttons[0] + msg->buttons[1];
In your opinion, do I also need to work on Joint jogging part? Thank you so much for your patience and kindness again. 

Steven

stevensu1838, 2018-11-13:
Hi Andy,

Today I tested your package with a real UR5 robot. I did it with both a Xbox360 and a space mouse. I could move the arm with both of the two devices. However, I firstly encountered error robot protective stops many times and the robot got stuck. I just couldn't jog the arm smoothly. Do you have any idea to improve it? Also, When I didn't meet the robot protective stop issues,  I still felt difficult to jog the arm in a straight line.(It is important Coz I am intending to do welding with UR5 robot.) Any suggestions on how I can control the TCP to go through a straight line? Is the pose of TCP also controlable?

Btw, do you have any publish paper or documents which can help me get a deeper and better understanding of this package? 
 Thanks a millions.

Gutesnacht, 2018-11-15:
Hi Steven,

I just comment on this Issue since we ran into similar problems yesterday. Nevertheless, we resolved them and are able to jog the robot (UR5) smoothly. 
Thank you very much for the jog_arm package and all your support @AndyZe ! I appreciate the effort you took to build that package! Wonderful work!

@stevensu1838 , To help you out, i am going to give you some of the details which we went through when installing the drivers and setting up the jog arm.

As Andy pointed out in [this](https://github.com/ros-industrial/ur_modern_driver/issues/219), you need the iron-ox ur modern driver.

Note- We build this on Ubuntu 16.04, Ros Kinetic and Version 3.5.10845 UR Polyscope 

1. Install ur_modern_driver, clone and build the iron-ox fork kinetic-devel branch of ur_modern_driver
     - When building the package, you will run into an error, that can be resolved through changes in the 
       in the ur_hardware_interface.cpp (from [here](https://github.com/iron-ox/ur_modern_driver/commit/883070d0b6c0c32b78bb1ca7155b8f3a1ead416c).
2. Install jog_arm and joy (you did that i guess)
     - Regarding joy, We had some problems due to outdated packages (ros), so sudo apt-get update and  
       sudo apt-get dist-upgrade resolved that, to be on the latest ros packages
3. Launch the driver with your robot ip (roslaunch ur_modern_driver ur5_bringup.launch robot_ip:=....)
4. Launch the MoveIt! config (roslaunch moveit_config ur5_moveit_planning_execution.launch)
5. Launch the jog_arm package - i guess you launched the spacenav mouse, so do that
6. Enjoy

Even though, most of this list was already mentioned by Andy, I thought i should condense this info to this topic to help out people that run into this problem.

The Key to get a smooth movement was to change to the iron-ox fork of the ur_modern_driver - before that, we had really jerky movements!

Hope this helps,

Best regards,

Lukas

stevensu1838, 2018-11-15:
Hi Lucas, 
Btw, can you please elaborate a bit what kind of changes I shall make to the interface.cpp file? Cheers

AndyZe, 2018-11-15:
Good to hear that others are having success.

@stevensu1838 your video looked good. You can turn off collision checking and make the parameters **lower_singularity_threshold** and **upper_singularity_threshold** really large if you want fewer pauses. You can make the parameter **incoming_command_timeout** larger, too. But, those changes can be a bit risky. That's not a ROS problem, that's an unavoidable physics problem.

Some benefits of ROS are:
-Singularity handling
-Collision detection
-Define and switch coordinate frames easily
-If you add more sensors later, that will be easier to do with ROS
-Community support

Gutesnacht, 2018-11-15:
Hi Steven,

You are welcomed to ask about our research. We currently use the jog_arm packet for teleportation via MARG-sensors (Magnetic Angularrate Gravity sensors). 
With theses sensors one can measure absolute orientation referenced to gravity and the geomagnetic field of the earth. This 3D orientation is than mapped to the robot in order to move it (search for AMiCUs - that was the first/previous system developed as part of the PhD Thesis from Nina Rudigkeit)

The changes you need to make to the interface are in the commit mentioned in my previous answer (replace red with the green indentation).

For the benefits of Ros, I agree with Andy's list.

Regards
Lukas

stevensu1838, 2018-11-19:
Hi Lukas, 

Thanks a lot for your help. I'd love to say your comments(`Regarding joy, We had some problems due to outdated packages (ros), so sudo apt-get update and sudo apt-get dist-upgrade resolved that, to be on the latest ros packages`)have been very helpful  in terms of joy node errors:```
 $ rosrun joy joy_node
[ INFO] [1541653009.395673744]: Opened joystick: /dev/input/js0. deadzone_: 0.050000.
Segmentation fault (core dumped)
```
I mention it here just in case anyone run into the same error again. 
Thanks a lot for sharing ideas of using Andy's package. Regarding this, may I please ask how did you get a deep understanding of jog_arm package? Like how the pack works and how you can do further development. I've been learning and using ROS and Moveit for a while but I am still a beginner. So hard to see through what's going on. I've found Andy's package very useful but I've love to know it well for further development. Can you please kindly share some of your ideas on it? Thanks a lit for your help. 

Cheers,
Steven

AndyZe, 2018-11-19:
Try to clean your catkin workspace, rebuild, then open a new terminal.

stevensu1838, 2018-11-30:
Hi Andy,
I'd love to say thank you for all the help that you've given me. I think with the above discussion everyone can use this package well. Now I've moved on to converting leapmotion message to twist in order to jog the arm. May I please ask if you've got your tuned UR5 Gazebo simulation? Coz my simulation still works so differently from the real robot. Thanks a million 

BR,
Steven

AndyZe, 2018-12-02:
@stevensu1838 i made a special simulation branch for you here (kinetic-steven) https://github.com/UTNuclearRobotics/ur5_sim/tree/kinetic-steven

The Readme has instructions. Make sure you're on the right branch when you clone it. This uses a SpaceNavigator because that's what I have available. If you need to change it for an XBox controller instead, see:

- ur_gazebo/controller/jog_settings.yaml

- ur_gazebo/launch/jog_arm.launch

stevensu1838, 2018-12-07:
Hi Andy, thank you so much for your help. I am still trying hard to use this special simulation. Will let you know when I make it. 
Can I please ask you **a question on use a LeapMotion sensor to jog the arm by modifying your spacenav_to_twist.cpp file**? I know you are also an expert of using Leap in ROS. I feel very sorry that I keep asking you questions but please believe me that I am trying hard myself. Thank you so much for your great support again. 
**My purpose is to use a hand to jog the ur5 arm in real time to do welding. such as when the palm goes to right in a line, the UR5 will weld a straight line. Palm pose maps to the pose of UR5's TCP and Palm orientation maps to the orientation of UR5's TCP.**  
Which is extremely like what this guy did in the pic:

[image]
To get the job done, I modified your spacenav_to_twist.cpp in the following way and I compile successfully. 
[image]
Please find the following the rqt_graph:
[image]
[image]
**Seems node leap_pub is talking to node spacenav_to_twist successfully. But I could not move the UR5 arm at all. Can you please give me some advice. Thanks a million for your great support.** 
For leap motion I am using leapros package from https://github.com/ros-drivers/leap_motion
BR,
Steven

stevensu1838, 2018-12-07:
Hi Andy,
My colleague suggested that I refer to your jog_api package. And make the following change in your code(**left: your code, right: after change**) 
[image]
Thanks a lot for your help. I am not a great programmer but I am on the way. If you can give me some direction, I am sure I can make it

AndyZe, 2018-12-07:
I don't want to get too involved in your research but it sounds like you're on a good track. What I would do is:

-In a loop, calculate the change in hand pose from the leap motion tracker
-check if the change is too large. Sometimes the leap motion loses track of the hand, so you need error checking for that
-apply some scaling to the change in pose, then send it as a jog command.

I'm going to close this issue now. Feel free to open a new one if a new problem pops up.

closed 2018-12-07 by AndyZe

stevensu1838, 2018-12-09:
Hi Andy,

I will keep it up. thank you again for your help. Cheers