UTNuclearRoboticsPublic/jog_arm #80: Can I please use jog_api package in Kinetic-devel branch of jog_arm?

stevensu1838, 2018-12-26:
Hi Andy,

First of all, I've achieved the control of UR5 with leap motion following your advice. Thanks with no end for your guidance. What's more, my colleague suggest that I should try out your jog_api package to control the UR5 robot and the reason is in jog_api Posestamped msg is used instead of Twiststamped msg.  He thinks the Posestamped way will be more reliable in terms of control the arm with human movements. 

Can you please kindly offer me any instructions on how to use your jog_api package?  The I can use the Posestamped method in it to control the arm. Thanks a million for your help. Hope you can visit New Zealand someday.

Cheers,
Steven

stevensu1838, 2019-04-05:
Hi Andy,

Just a quick question on your jog_api. I talked with an experienced ROS user about the idea of using Moveit to do point-to-point motion, like sending the robot from A to B. And he replied as below: 

_You mentioned that you want to joint several trajectories together into a single smooth movement. The main challenge with this is that as standard MoveiIt will project trajectories who's start and end velocities are zero, this mean that if you join them together the motion will 'stutter' as the arm stops and starts between trajectories._

My question is how is your jog_api different from the Moveit python/C++ API in terms of moving the the TCP through some certain points? Will your jog_api avoid the 'stutter' issue described by above? Thanks a million.

BR,
Steven

AndyZe, 2019-04-05:
Yes, it will avoid the stuttering with a low-pass filter. You can adjust the filter in the config file.

There are better ways of doing it, but I don't think I can release that to the public right now.