UTNuclearRoboticsPublic/jog_arm #86: Invalid argument passed to lookupTransform argument source_frame in tf2 frame_ids cannot be empty

AshwinR96, 2019-02-09:
I am trying to run jog_arm to control my UR3 simulation on Gazebo. But I am getting this error:

>/jog_arm_server: Invalid argument passed to lookupTransform argument source_frame in tf2 frame_ids cannot be empty

What do I need to do fix this?

My jog_settings.yaml has this:
`gazebo: true # Whether the robot is started in a Gazebo simulation environment
collision_check: true # Check collisions?
command_in_type: "unitless" # "unitless"> in the range [-1:1], as if from joystick. "speed_units"> cmds are in m/s and rad/s
scale: # Only used if command_in_type=="unitless"
  linear:  0.003  # Max linear velocity. Meters per publish_period.
  rotational:  0.006  # Max angular velocity. Rads per publish_period.
  joint: 0.01  # Max joint angular/linear velocity. Rads or Meters per publish period.
cartesian_command_in_topic:  jog_arm_server/delta_jog_cmds # Topic for xyz commands
joint_command_in_topic: jog_arm_server/joint_delta_jog_cmds # Topic for angle commands
command_frame:  base_link  # TF frame that incoming cmds are given in
incoming_command_timeout:  5  # Stop jogging if X seconds elapse without a new cmd
joint_topic:  joint_states
move_group_name:  manipulator  # Often 'manipulator' or 'arm'
lower_singularity_threshold:  30  # Start decelerating when the condition number hits this (close to singularity). Larger --> closer to singularity
hard_stop_singularity_threshold: 45 # Stop when the condition number hits this. Larger --> closer to singularity
lower_collision_proximity_threshold: 0.1 # Start decelerating when a collision is this far [m]
hard_stop_collision_proximity_threshold: 0.005 # Stop when a collision is this far [m]
planning_frame: base_link  # The MoveIt! planning frame. Often 'base_link'
low_pass_filter_coeff: 2.  # Larger-> more smoothing to jog commands, but more lag.
publish_period: 0.008  # 1/Nominal publish rate [seconds]
publish_delay: 0.005  # delay between calculation and execution start of command
collision_check_rate: 5 # [Hz] Collision-checking can easily bog down a CPU if done too often.
# Publish boolean warnings to this topic
warning_topic: jog_arm_server/warning
joint_limit_margin: 0.1 # added as a buffer to joint limits [radians]. If moving quickly, make this larger.
command_out_topic: arm_controller/command
# What type of topic does your robot driver expect?
# Currently supported are std_msgs/Float64MultiArray (for ros_control JointGroupVelocityController)
# or trajectory_msgs/JointTrajectory (for Universal Robots and other non-ros_control robots)
command_out_type: trajectory_msgs/JointTrajectory
# Can save some bandwidth as most robots only require positions or velocities
publish_joint_positions: true
publish_joint_velocities: true
publish_joint_accelerations: false`

AndyZe, 2019-02-09:
You just need to make sure the incoming message has a header.stamp.frame_id. I'll push a fix later today

AshwinR96, 2019-02-09:
Okay, thank you

AndyZe, 2019-02-09:
Commit 13ad62a should take care of this. The `command_frame` from your yaml file is automatically put on all incoming msgs. Let me know if it still isn't working for you.

closed 2019-02-09 by AndyZe