UTNuclearRoboticsPublic/jog_arm #91: locate the jog_arm package.

Kevinlinpr, 2019-03-07:
Hi there,I can not locate the jog_arm package.
I execute the command **sudo apt install ros-kinetic-joy** then I get the return from console

> _**Reading package lists... Done
> Building dependency tree       
> Reading state information... Done
> ros-kinetic-joy is already the newest version (1.12.0-0xenial-20181107-011933-0800).
> 0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.**_

through above log, I know the ros-kinetic-joy is already installed while I install the ros-kinetic.
According to [your document about the package](http://wiki.ros.org/jog_arm),

> Clone and build this package. 

I can not locate this package and I wonder how I can find this package and build it?
Which search-tool can I use to find it and which build-tool can I use to build this source package?
Thanks!

Kevinlinpr, 2019-03-07:
And I try to create a new ros workspace to make a test ,then I extract the compress file **jog_arm-kinetic.zip** and put the entire directory into the new workspace and go into this path,finally make a build command **catkin_make** and I get the below error.

```
Base path: /home/kevinlinpr/Public/realtime_control
Source space: /home/kevinlinpr/Public/realtime_control/src
Build space: /home/kevinlinpr/Public/realtime_control/build
Devel space: /home/kevinlinpr/Public/realtime_control/devel
Install space: /home/kevinlinpr/Public/realtime_control/install
####
#### Running command: "cmake /home/kevinlinpr/Public/realtime_control/src -DCATKIN_DEVEL_PREFIX=/home/kevinlinpr/Public/realtime_control/devel -DCMAKE_INSTALL_PREFIX=/home/kevinlinpr/Public/realtime_control/install -G Unix Makefiles" in "/home/kevinlinpr/Public/realtime_control/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/kevinlinpr/Public/realtime_control/devel
-- Using CMAKE_PREFIX_PATH: /home/kevinlinpr/Public/robot_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/kevinlinpr/Public/robot_ws/devel;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/kevinlinpr/Public/realtime_control/build/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.14
-- BUILD_SHARED_LIBS is on
WARNING: Package name "kevinlinprV4" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 2 packages in topological order:
-- ~~  - jog_msgs
-- ~~  - jog_arm
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'jog_msgs'
-- ==> add_subdirectory(jog_arm-kinetic/jog_msgs)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- jog_msgs: 2 messages, 0 services
-- +++ processing catkin package: 'jog_arm'
-- ==> add_subdirectory(jog_arm-kinetic/jog_arm)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by
  "rosparam_shortcuts" with any of the following names:

    rosparam_shortcutsConfig.cmake
    rosparam_shortcuts-config.cmake

  Add the installation prefix of "rosparam_shortcuts" to CMAKE_PREFIX_PATH or
  set "rosparam_shortcuts_DIR" to a directory containing one of the above
  files.  If "rosparam_shortcuts" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  jog_arm-kinetic/jog_arm/CMakeLists.txt:8 (find_package)

-- Could not find the required component 'rosparam_shortcuts'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "rosparam_shortcuts" with any of the following names:

    rosparam_shortcutsConfig.cmake
    rosparam_shortcuts-config.cmake

  Add the installation prefix of "rosparam_shortcuts" to CMAKE_PREFIX_PATH or
  set "rosparam_shortcuts_DIR" to a directory containing one of the above
  files.  If "rosparam_shortcuts" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  jog_arm-kinetic/jog_arm/CMakeLists.txt:8 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/kevinlinpr/Public/realtime_control/build/CMakeFiles/CMakeOutput.log".
See also "/home/kevinlinpr/Public/realtime_control/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

```

AndyZe, 2019-03-07:
Sorry for your troubles. Please install rosparam_shortcuts:

sudo apt install ros-kinetic-rosparam-shortcuts

Kevinlinpr, 2019-03-08:
It works on me.Thank you so much!

closed 2019-03-08 by Kevinlinpr