UTNuclearRoboticsPublic/jog_arm #94: Connecting jog_arm to real UR3

AshwinR96, 2019-03-20:
HI,

I am trying to use jog_arm with the real ur3 but i am not able to send commands to the real robot. I was able to connect jog_arm to a gazebo simulation. I have listed below all my config files, rostopic list and roslaunch file. Can you please help me find out what the issue is?

I am not sure what the correct controller settings are. I tried shutting down all controllers except joint_group_vel_controller and my config files are set like this:

My jog_settings.yaml file is:
`gazebo: false # Whether the robot is started in a Gazebo simulation environment
collision_check: false # Check collisions?
command_in_type: "unitless" # "unitless"> in the range [-1:1], as if from joystick. "speed_units"> cmds are in m/s and rad/s
scale: # Only used if command_in_type=="unitless"
  linear:  0.004  # Max linear velocity. Meters per publish_period. Units is [m/s]
  rotational:  0.008  # Max angular velocity. Rads per publish_period. Units is [rad/s]
  joint: 0.003  # Max joint angular/linear velocity. Rads or Meters per publish period. Units is [rad/s] or [m/s].
cartesian_command_in_topic: jog_arm_server/delta_jog_cmds  # Topic for xyz commands
joint_command_in_topic: jog_arm_server/joint_delta_jog_cmds # Topic for angle commands
command_frame:  base_link  # TF frame that incoming cmds are given in
incoming_command_timeout:  0.2  # Stop jogging if X seconds elapse without a new cmd
joint_topic:  joint_states
move_group_name:  manipulator  # Often 'manipulator' or 'arm'
lower_singularity_threshold:  30  # Start decelerating when the condition number hits this (close to singularity)
hard_stop_singularity_threshold: 45 # Stop when the condition number hits this
lower_collision_proximity_threshold: 0.1 # Start decelerating when a collision is this far [m]
hard_stop_collision_proximity_threshold: 0.005 # Stop when a collision is this far [m]
planning_frame: base_link  # The MoveIt! planning frame. Often 'base_link'
low_pass_filter_coeff: 2.  # Larger --> trust the filtered data more, trust the measurements less.
publish_period: 0.008  # 1/Nominal publish rate [seconds]
publish_delay: 0.005  # delay between calculation and execution start of command
collision_check_rate: 5 # [Hz] Collision-checking can easily bog down a CPU if done too often.
# Publish boolean warnings to this topic
warning_topic: jog_arm_server/warning
joint_limit_margin: 0.1 # added as a buffer to joint limits [radians]. If moving quickly, make this larger.
command_out_topic: joint_group_vel_controller/command
# What type of topic does your robot driver expect?
# Currently supported are std_msgs/Float64MultiArray (for ros_control JointGroupVelocityController)
# or trajectory_msgs/JointTrajectory (for Universal Robots and other non-ros_control robots)
command_out_type: std_msgs/Float64MultiArray
# Can save some bandwidth as most robots only require positions or velocities
publish_joint_positions: false
publish_joint_velocities: true
publish_joint_accelerations: false
`

and my ur3_moveit_config controllers.yaml file is (this actually gives me an error saying unknown controller type "velocity_controllers/JointGroupVelocityController") :

`controller_list:
  - name: ""
    action_ns: joint_group_vel_controller/command
    type: velocity_controllers/JointGroupVelocityController
    joints:
      - shoulder_pan_joint
      - shoulder_lift_joint
      - elbow_joint
      - wrist_1_joint
      - wrist_2_joint
      - wrist_3_joint`

when i do rostopic list i get:

`/attached_collision_object
/clicked_point
/collision_object
/execute_trajectory/cancel
/execute_trajectory/feedback
/execute_trajectory/goal
/execute_trajectory/result
/execute_trajectory/status
/initialpose
/jog_arm_server/delta_jog_cmds
/jog_arm_server/joint_delta_jog_cmds
/jog_arm_server/warning
/joint_group_vel_controller/command
/joint_states
/joint_temperature
/move_base_simple/goal
/move_group/cancel
/move_group/display_contacts
/move_group/display_planned_path
/move_group/feedback
/move_group/goal
/move_group/monitored_planning_scene
/move_group/ompl/parameter_descriptions
/move_group/ompl/parameter_updates
/move_group/plan_execution/parameter_descriptions
/move_group/plan_execution/parameter_updates
/move_group/planning_scene_monitor/parameter_descriptions
/move_group/planning_scene_monitor/parameter_updates
/move_group/result
/move_group/sense_for_plan/parameter_descriptions
/move_group/sense_for_plan/parameter_updates
/move_group/status
/move_group/trajectory_execution/parameter_descriptions
/move_group/trajectory_execution/parameter_updates
/pickup/cancel
/pickup/feedback
/pickup/goal
/pickup/result
/pickup/status
/place/cancel
/place/feedback
/place/goal
/place/result
/place/status
/planning_scene
/planning_scene_world
/recognized_object_array
/rosout
/rosout_agg
/rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/feedback
/rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/update
/rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/update_full
/rviz_parallels_vm_7426_4186558235624850098/motionplanning_planning_scene_monitor/parameter_descriptions
/rviz_parallels_vm_7426_4186558235624850098/motionplanning_planning_scene_monitor/parameter_updates
/tf
/tf_static
/tool_velocity
/trajectory_execution_event
/ur_driver/URScript
/ur_driver/io_states
/ur_driver/robot_status
/wrench
`

My roslaunch file is ( I took out the ip address value):

`<launch> 

<include file="$(find ur_modern_driver)/launch/ur3_ros_control.launch">
    <arg name="robot_ip" value=""/>
  </include>

  <include file="$(find ur3_moveit_config)/launch/ur3_moveit_planning_execution.launch">
		<arg name="sim" value="false"/>    
		<arg name="limited" value="true"/>
  </include>

  <include file="$(find ur3_moveit_config)/launch/moveit_rviz.launch">
    <arg name="config" value="true"/>
  </include> 

   <node name="jog_arm_server" pkg="jog_arm" type="jog_arm_server" output="screen" required="true" >
    <param name="parameter_ns" type="string" value="jog_arm_server" />
    <rosparam command="load" file="$(find jog_arm)/config/jog_settings.yaml" />
  </node>

  <node pkg="move_ur3" type="realrobot.py" name="teleop"  output="screen" required="true">
  </node>

</launch>`

AndyZe, 2019-03-20:
My quick initial guess is that you shouldn't have killed the joint_states controller. It publishes joints from the robot driver, I think. Try re-launching and stopping only the default trajectory controller.

AndyZe, 2019-03-20:
I know this ros_control stuff can be confusing...

AshwinR96, 2019-03-20:
Hi Andy thank you very much for the quick response, I added the joint_states controller back in could you also help me with the ur3_moveit_config controllers.yaml file? What is the correct controller type?

My yaml file is like this at the moment:

`controller_list:

name: ""
action_ns: joint_group_vel_controller/command
type: velocity_controllers/JointGroupVelocityController
joints:
shoulder_pan_joint
shoulder_lift_joint
elbow_joint
wrist_1_joint
wrist_2_joint
wrist_3_joint`

AndyZe, 2019-03-20:
This is what I have. There shouldn't be an action_ns because there's no action associated with this controller.

joint_group_vel_controller:
   type: velocity_controllers/JointGroupVelocityController
   joints:
     - shoulder_pan_joint
     - shoulder_lift_joint
     - elbow_joint
     - wrist_1_joint
     - wrist_2_joint
     - wrist_3_joint

AshwinR96, 2019-03-20:
I set my ur3_moveit_config like this:
[image]

and i get the error

[ERROR] [1553114876.935109804]: Unknown controller type: velocity_controllers/JointGroupVelocityController

AndyZe, 2019-03-20:
try
sudo apt install ros-kinetic-ros-contro*

AshwinR96, 2019-03-20:
I am still getting the same error after installing that

AndyZe, 2019-03-20:
You may need to re-load the controller library. See the wiki here.

http://wiki.ros.org/controller_manager

rosservice call /controller_manager/reload-libraries

or something like that

AshwinR96, 2019-03-20:
I ran that:

[image]

but i still get the same error when i run move_group.launch:
[image]

AshwinR96, 2019-03-21:
I am actually able to control the ur3 using jog_arm now, I just ignored this unknown controller type error. 

In jog_arm I am set the command out topic to ur_driver/joint_states, and this worked.

Thanks for your help.

closed 2019-03-21 by AshwinR96

AndyZe, 2019-03-21:
Hmm, that's a new and interesting way to do it

AndyZe, 2019-03-21:
I think the confusion arose because most people are using ur_modern_driver these days, not ur_driver

Tina1994, 2019-05-31:
> I am actually able to control the ur3 using jog_arm now, I just ignored this unknown controller type error.
> 
> In jog_arm I am set the command out topic to ur_driver/joint_states, and this worked.
> 
> Thanks for your help.

Tina1994, 2019-05-31:
@AshwinR96  Did you solve this problem finally? I meet the same question.
       And you mentioned that you ignored the error and still can do the velocity control? Is it true?
       Hope for your help, thanks very much.

AshwinR96, 2019-05-31:
@Tina1994 Yeah i was able to do velocity control, my understanding is that jog_arm doesnt need moveit's controllers to move the robot and MoveIt! starts after a few seconds despite giving the error message. I used the ur_modern_driver package to open the joint_states topic,  and set jog_arm publish to publish to the joint_speeds topic. 

The joint_states topic is launched with this:
`roslaunch ur_modern_driver ur3_bringup.launch robot_ip:=IP_OF_THE_ROBOT'

This is my jog_settings.yaml file:

`gazebo: false # Whether the robot is started in a Gazebo simulation environment
collision_check: true # Check collisions?
command_in_type: "unitless" # "unitless"> in the range [-1:1], as if from joystick. "speed_units"> cmds are in m/s and rad/s
scale: # Only used if command_in_type=="unitless"
  linear:  0.0001  # Max linear velocity. Meters per publish_period.
  rotational:  0.0015  # Max angular velocity. Rads per publish_period.
  joint: 0.0015  # Max joint angular/linear velocity. Rads or Meters per publish period.
cartesian_command_in_topic:  jog_arm_server/delta_jog_cmds # Topic for xyz commands
joint_command_in_topic: jog_arm_server/joint_delta_jog_cmds # Topic for angle commands
command_frame:  base_link  # TF frame that incoming cmds are given in
incoming_command_timeout:  0.5  # Stop jogging if X seconds elapse without a new cmd
joint_topic:  joint_states
move_group_name:  manipulator  # Often 'manipulator' or 'arm'
lower_singularity_threshold:  30  # Start decelerating when the condition number hits this (close to singularity). Larger --> closer to singularity
hard_stop_singularity_threshold: 45 # Stop when the condition number hits this. Larger --> closer to singularity
lower_collision_proximity_threshold: 0.1 # Start decelerating when a collision is this far [m]
hard_stop_collision_proximity_threshold: 0.005 # Stop when a collision is this far [m]
planning_frame: base_link  # The MoveIt! planning frame. Often 'base_link'
low_pass_filter_coeff: 12.  # Larger-> more smoothing to jog commands, but more lag.
publish_period: 0.008  # 1/Nominal publish rate [seconds]
publish_delay: 0.005  # delay between calculation and execution start of command
collision_check_rate: 5 # [Hz] Collision-checking can easily bog down a CPU if done too often.
# Publish boolean warnings to this topic
warning_topic: jog_arm_server/warning
joint_limit_margin: 0.1 # added as a buffer to joint limits [radians]. If moving quickly, make this larger.
#command_out_topic: ur_driver/joint_speed # added as a buffer to joint limits [radians]. If moving quickly, make this larger.
command_out_topic: ur_driver/joint_speed
# What type of topic does your robot driver expect?
# Currently supported are std_msgs/Float64MultiArray (for ros_control JointGroupVelocityController)
# or trajectory_msgs/JointTrajectory (for Universal Robots and other non-ros_control robots)
command_out_type: trajectory_msgs/JointTrajectory
# Can save some bandwidth as most robots only require positions or velocities
publish_joint_positions: true
publish_joint_velocities: true
publish_joint_accelerations: false`

gavanderhoorn, 2019-05-31:
@AshwinR96 wrote:
> I am actually able to control the ur3 using jog_arm now, I just ignored this unknown controller type error.
> 
> In jog_arm I am set the command out topic to ur_driver/joint_states, and this worked.

that cannot work. The `joint_states` topic only broadcasts current state, it doesn't accept any commands.