UTNuclearRoboticsPublic/jog_arm #95: How do I control the the velocity precisely or set the velocity a certain value?

stevensu1838, 2019-04-03:
Hi all,

Thank you so much for your great package. It has been very helpful. I am using jog arm for welding and I've got a question on how to control the the velocity precisely. Because I want to make the TCP travel around 5 mm/second. I've noticed that the max and min velocity can be set in the config file but is there a way I can fix the velocity? Also, is there any way I can change the setting of the velocity through code instead of simply using the config file? I mean I want to make it change during the process of jogging.

Secondly, may I please ask if it is possible that I use your jog arm server to subscribe to two nodes. One is the normal 3D mouse node to jog the arm. Another one is a sensor node that detect the distance from the TCP to the working piece. When the TCP of UR5 is closer than 20 cm, the velocity in Z direction remain 0. At the same time, I jog the arm in X and Y directions to conduct welding task. 

The reason why I got this idea is for welding you always want you robot TCP to keep a certain distance to the working piece. If they are too close, you crash you welding gun or the working piece. 

Thank you again for giving me hope on my project. Looking forward to hearing from you soon

BR,

Steven

AndyZe, 2019-04-04:
Hi Steven,

See this line in the config file:

`command_in_type: "unitless" # "unitless"> in the range [-1:1], as if from joystick. "speed_units"> cmds are in m/s and rad/s`

As it says, if you change that line to "speed_units", it interprets incoming commands as m/s and rad/s speeds. So that's how you can set velocity directly.

You can take input from two nodes easily. I would suggest looking at the twist_mux package. It lets you give a priority to commands from one topic over another.

AndyZe, 2019-04-04:
Feel free to re-open if you have related issues. Closing for now.

closed 2019-04-04 by AndyZe

stevensu1838, 2019-04-07:
Hi Andy,

I'd like to first say thank you for your constant help.
Changing "unitless" to "speed_units" has made me able to control the velocity of TCP in simulation with the following commands. 
Terminal1: roslaunch ur_gazebo temoto_ur5.launch limited:=true
Terminal2: roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch sim:=true limited:=true
(Optional)Another: roslaunch ur5_moveit_config moveit_rviz.launch config:=true
Terminal3: roslaunch ur_gazebo jog_arm.launch

When it comes to the velocity commands, I am not using a joystick or 3D mouse. What I do is I move a cude in Unity(3D game engine) and convert the velocity of this cube to Twist message. Eventually , this Twist message is assigned the Twiststamped msg in spacenav_to_twist.cpp in the Jog arm package. In this case, I expected that the TCP of UR5 would copy the movement of the 3D cube that I move in 3D in Unity. Please see the rqt_graph below.

[image]

So far I've been getting there and the only issue is after I successfully jog the arm for a few seconds, everything freezes like the pic below and it doesn't allow me to move the arm any more. Can I please show you my yaml file and you take a look at these settings?  The main problem for now is I can jog the arm which changing velocity commands but the jogging doesn't last long. Apart from the yaml setting, do you see any other causes? Thank you so much for your support. Really appreciated.

[image]

Please see below my jog_settings.yaml file under directory /home/mario/catkin_ws/src/ur5_sim/ur_gazebo/controller
gazebo: true # Whether the robot is started in a Gazebo simulation environment
allow_dropped_DOF: true # Allow dropping a DOF to skirt a singularity?
collision_check: false # Check collisions?
command_in_type: "speed_units" # "unitless"> in the range [-1:1], as if from joystick. "speed_units"> cmds are in m/s and rad/s #### Steven changed from unitless to speed_units 05042019
scale: # Only used if command_in_type=="unitless"
  linear:  0.010  # Max linear velocity. Meters per publish_period. Units is [m/s]
  rotational:  0.010  # Max angular velocity. Rads per publish_period. Units is [rad/s]
  joint: 0.05  # Max joint angular/linear velocity. Rads or Meters per publish period. Units is [rad/s] or [m/s].
cartesian_command_in_topic:  jog_arm_server/delta_jog_cmds # Topic for xyz commands
joint_command_in_topic: jog_arm_server/joint_delta_jog_cmds # Topic for angle commands
command_frame:  base_link  # TF frame that incoming cmds are given in
incoming_command_timeout:  5  # Stop jogging if X seconds elapse without a new cmd
joint_topic:  joint_states
move_group_name:  manipulator  # Often 'manipulator' or 'arm'
lower_singularity_threshold:  50  # Start decelerating when the condition number hits this (close to singularity)
hard_stop_singularity_threshold: 100 # Stop when the condition number hits this
lower_collision_proximity_threshold: 0.1 # Start decelerating when a collision is this far [m]
hard_stop_collision_proximity_threshold: 0.005 # Stop when a collision is this far [m]
command_out_topic: arm_controller/command
planning_frame: base_link  # The MoveIt! planning frame. Often 'base_link'
low_pass_filter_coeff: 2.  # Larger --> trust the filtered data more, trust the measurements less.
publish_period: 0.008  # 1/Nominal publish rate [seconds]
publish_delay: 0.005  # delay between calculation and execution start of command
# Publish boolean warnings to this topic
warning_topic: jog_arm_server/warning
joint_limit_margin: 0.1 # added as a buffer to joint limits [radians]. If moving quickly, make this larger.
# What type of topic does your robot driver expect?
# Currently supported are std_msgs/Float64MultiArray (for ros_control JointGroupVelocityController)
# or trajectory_msgs/JointTrajectory (for Universal Robots and other non-ros_control robots)
command_out_type: trajectory_msgs/JointTrajectory
# Can save some bandwidth as most robots only require positions or velocities
publish_joint_positions: true
publish_joint_velocities: false
publish_joint_accelerations: false

Thank you again for all the guidance and help that you've given me. It is you who give me hope. 

BR,
Steven

stevensu1838, 2019-04-07:
Hi Andy,

Just want to let you know that when the **robot in Gazebo** and **the console displaying dropping trajectory points**(the fist console you see in the screenshot above)  freeze(stop running), the jog_arm_server/delta_jog_cmds topic is still working as normal.
 
I did rostopic echo to ensure that and it printed the twist values in X,Y,Z as noraml. Hopefully, you can help me out.

stevensu1838, 2019-04-10:
Hi Andy,
Please take a look at the video clip of my problem after switching to speed_unit. The jogging doesn't last long.
[https://drive.google.com/file/d/1TraAR9h--d0biwMyuCkUvZSl91JnPCbJ/view?usp=sharing](url)

stevensu1838, 2019-04-11:
Hi Andy,
If possible, can you please take a look at this issue. I don't use ROS well enough to debug this error. And I can ensure this issue(freezing) happens only when I switch form unitless to speed_unit. Not about how I sent the commands to jog_arm_server. I am using the ur_gazebo provided by you. Should be reliable. You experienced this before?

I will test in on the real UR5 tomorrow. Hope I get luck. I am growing up really poor and I don't have a way out but to finish this project to get my certificate. Thanks a million.

stevensu1838, 2019-04-12:
Hi Andy,

Just wanna let you know in  speed_unit mode, If I keep the velocity in x,y and z under 0.05 mm/sec. This issue doesn't happen again and I can jog the arm as long as I want. It is very slow but meets my requirements for welding.  Just curious if the highest speed can be modified or not. Anyways, thanks a lot again. Will show you around when you come to NZ someday.

Abduoit, 2020-07-25:
Hi @stevensu1838 I read the entire discussion, and I have question please.

I am trying to jog UR5 with gazebo using ROS indigo (I can't move to kinetic at the moment because I need to use the manipulator with Husky mobile robot which only works on indigo).  

The `jog_arm_server/delta_jog_cmds` topic  works as normal.
I set command output topic like this ` cmd_out_topic: arm_controller/command` 

Controller list in gazebo like that:
```
abdu@abdu-uoit:~/devel_ws$ rosservice call /controller_manager/list_controllers
controller: 
  - 
    name: joint_state_controller
    state: running
    type: joint_state_controller/JointStateController
    hardware_interface: hardware_interface::JointStateInterface
    resources: []
  - 
    name: arm_controller
    state: running
    type: position_controllers/JointTrajectoryController
    hardware_interface: hardware_interface::PositionJointInterface
    resources: ['elbow_joint', 'shoulder_lift_joint', 'shoulder_pan_joint', 'wrist_1_joint', 'wrist_2_joint', 'wrist_3_joint']
```

Everything seems normal, but I still get the following error when I start jogging!! I appreciate any help? 

`Client [/gazebo] wants topic /arm_controller/command to have datatype/md5sum [trajectory_msgs/JointTrajectory/65b4f94a94d1ed67169da35a02f33d3f], but our version has [std_msgs/String/992ce8a1687cec8c8bd883ec73ca41d1]. Dropping connection.`

Gutesnacht, 2020-07-25:
Hey Abaduit,you should switch to a joint_velocity_group controller. That should hopefully fix your problem.

Abduoit, 2020-07-25:
@Gutesnacht Thanks, but I tried to add `velocity_controller` to `arm_controller_ur5.yaml` as seen below, but it didn't work, Can you please explain how ?   

```
joint_group_position_controller:
  type: velocity_controllers/JointGroupVelocityController
  joints:
     - shoulder_pan_joint
     - shoulder_lift_joint
     - elbow_joint
     - wrist_1_joint
     - wrist_2_joint
     - wrist_3_joint
```

Abduoit, 2020-07-26:
I switched to joint_velocity, after modifying all `<hardwareInterface>VelocityJointInterface</hardwareInterface>` in file `ur.transmission.xacro`

This is `arm_controller_ur5.yaml` in `ur_gazebo/controller`

```
arm_controller:
  type: velocity_controllers/JointTrajectoryController
  joints:
     - shoulder_pan_joint
     - shoulder_lift_joint
     - elbow_joint
     - wrist_1_joint
     - wrist_2_joint
     - wrist_3_joint
  constraints:
      goal_time: 0.6
      stopped_velocity_tolerance: 0.05
      shoulder_pan_joint: {trajectory: 0.1, goal: 0.1}
      shoulder_lift_joint: {trajectory: 0.1, goal: 0.1}
      elbow_joint: {trajectory: 0.1, goal: 0.1}
      wrist_1_joint: {trajectory: 0.1, goal: 0.1}
      wrist_2_joint: {trajectory: 0.1, goal: 0.1}
      wrist_3_joint: {trajectory: 0.1, goal: 0.1}
  stop_trajectory_duration: 0.5
  state_publish_rate:  125
  action_monitor_rate: 10
  gains:
    #!!These values are useable, but maybe not optimal
    shoulder_pan_joint: {p: 1.2,  i: 0.0, d: 0.1, i_clamp: 1}
    shoulder_lift_joint: {p: 1.2,  i: 0.0, d: 0.1, i_clamp: 1}
    elbow_joint: {p: 1.2,  i: 0.0, d: 0.1, i_clamp: 1}
    wrist_1_joint: {p: 1.2,  i: 0.0, d: 0.1, i_clamp: 1}
    wrist_2_joint: {p: 1.2,  i: 0.0, d: 0.1, i_clamp: 1}
    wrist_3_joint: {p: 1.2,  i: 0.0, d: 0.1, i_clamp: 1}
```

list_controllers as shown 

```
abdu@abdu-uoit:~/devel_ws$ rosservice call /controller_manager/list_controllers
controller: 
  - 
    name: arm_controller
    state: running
    type: velocity_controllers/JointTrajectoryController
    hardware_interface: hardware_interface::VelocityJointInterface
    resources: ['elbow_joint', 'shoulder_lift_joint', 'shoulder_pan_joint', 'wrist_1_joint', 'wrist_2_joint', 'wrist_3_joint']
  - 
    name: joint_state_controller
    state: running
    type: joint_state_controller/JointStateController
    hardware_interface: hardware_interface::JointStateInterface
    resources: []
```

jog yaml file 

```
jog_arm_server:
  simu: true # Whether the robot is started in simulation environment
  gazebo: true # Whether the robot is started in a Gazebo simulation environment
  collision_check: true # Check collisions?
  cmd_in_type: "unitless" # "unitless"> in the range [-1:1], as if from joystick. "speed_units"> cmds are in m/s and rad/s
  scale: # Only used if command_in_type=="unitless"
    linear:  0.004  # Max linear velocity. Meters per publish_period.
    rotational:  0.008  # Max angular velocity. Rads per publish_period.
    joint: 0.05  # Max joint angular/linear velocity. Rads or Meters per publish period.

# topics
  cmd_in_topic:  jog_arm_server/delta_jog_cmds # Topic for xyz commands
  joint_command_in_topic: jog_arm_server/joint_delta_jog_cmds # Topic for angle commands

  cmd_frame: ee_link   # TF frame that incoming cmds are given in

  incoming_cmd_timeout:  20  # Stop jogging if X seconds elapse without a new cmd
  joint_topic:  joint_states

  move_group_name:  manipulator  # Often 'manipulator' or 'arm'

  singularity_threshold:  30  # Start decelerating when the condition number hits this (close to singularity). Larger --> closer to singularity
  hard_stop_singularity_threshold: 45 # Stop when the condition number hits this. Larger --> closer to singularity
  lower_collision_proximity_threshold: 0.1 # Start decelerating when a collision is this far [m]
  hard_stop_collision_proximity_threshold: 0.005 # Stop when a collision is this far [m]

  planning_frame: base_link  # The MoveIt! planning frame. Often 'base_link'

  low_pass_filter_coeff: 2.  # Larger-> more smoothing to jog commands, but more lag.
#publish_period: 0.008  # 1/Nominal publish rate [seconds]
  pub_period: 0.01  # 1/Nominal publish rate [seconds]
  publish_delay: 0.005  # delay between calculation and execution start of command
  collision_check_rate: 5 # [Hz] Collision-checking can easily bog down a CPU if done too often.

# Publish boolean warnings to this topic
  warning_topic: jog_arm_server/warning
  joint_limit_margin: 0.1 # added as a buffer to joint limits [radians]. If moving quickly, make this larger.

  cmd_out_topic: arm_controller/command

# What type of topic does your robot driver expect?

# Currently supported are std_msgs/Float64MultiArray (for ros_control JointGroupVelocityController)
# or trajectory_msgs/JointTrajectory (for Universal Robots and other non-ros_control robots)

  cmd_out_type: std_msgs/Float64MultiArray
  #cmd_out: trajectory_msgs/JointTrajectory

# Can save some bandwidth as most robots only require positions or velocities
  publish_joint_positions: true
  publish_joint_velocities: true
  publish_joint_accelerations: false
```

But I still have same error, Any help please ??

`[ERROR] [1595784022.754663767, 338.189000000]: Client [/gazebo] wants topic /arm_controller/command to have datatype/md5sum [trajectory_msgs/JointTrajectory/65b4f94a94d1ed67169da35a02f33d3f], but our version has [std_msgs/String/992ce8a1687cec8c8bd883ec73ca41d1]. Dropping connection.`