UTNuclearRoboticsPublic/jog_arm #100: compare the velocity commands sending to jog arm and the actually velocity of UR5 TCP

stevensu1838, 2019-05-09:
Hi Andy，
As you know I replaced the space mouse with a motion sensor which can track the velocity of user's hand.  

My supervisor suggested that I compare the velocity commands sending to jog arm and the actually velocity of UR5 TCP. Thus, we can check out if the human motion is exactly mapped to the TCP of UR5. If so, I can move onto other tasks. If not, I might you some control methods(like predictive control) to improve it.

At the moment, I am thinking of plotting both the velocity commands from then motion sensor and the actually velocity of UR5 TCP. And I reckon I can simply print out the incoming velocity commands by doing rosplot echo /leapmotion_to_twist.

My question is How about the real velocity of the robot TCP? Is there a topic that I can use to represent the actual speed? If you were me, how would you obtain the real velocity of TCP for compaction?

If you're interested, A reference on using predictive control to improve the velocity control of TCP of UR5
[https://ieeexplore.ieee.org/document/6699723](url)

stevensu1838, 2019-05-28:
Hi all,
For any of you who wants to read data in real-time from the TCP of UR5, please check out the following site:
https://www.universal-robots.com/how-tos-and-faqs/how-to/ur-how-tos/real-time-data-exchange-rtde-guide-22229/

To read the data, please run the record.py code as shown below:
import argparse
import logging
import sys
sys.path.append('..')
import rtde.rtde as rtde
import rtde.rtde_config as rtde_config
import rtde.csv_writer as csv_writer

DELIMITER = ","

#parameters
parser = argparse.ArgumentParser()
parser.add_argument('--host', default='192.168.1.1', help='name of host to connect to (localhost)')
parser.add_argument('--port', type=int, default=30004, help='port number (30004)')
parser.add_argument('--samples', type=int, default=0, help='number of samples to record')
parser.add_argument('--frequency', type=int, default=125, help='the sampling frequency in Herz')
parser.add_argument('--config', default='pose_vel_logger_config.xml', help='data configuration file to use (record_configuration.xml)')
parser.add_argument('--output', default='robot_data.csv', help='data output file to write to (robot_data.csv)')
parser.add_argument("--verbose", help="increase output verbosity", action="store_true")
args = parser.parse_args()

if args.verbose:
    logging.basicConfig(level=logging.INFO)

conf = rtde_config.ConfigFile(args.config)
output_names, output_types = conf.get_recipe('out')

con = rtde.RTDE(args.host, args.port)
con.connect()

# get controller version
con.get_controller_version()

# setup recipes
if not con.send_output_setup(output_names, output_types, frequency = args.frequency):
    logging.error('Unable to configure output')
    sys.exit()

#start data synchronization
if not con.send_start():
    logging.error('Unable to start synchronization')
    sys.exit()

with open(args.output, 'w') as csvfile:
    writer = csv_writer.CSVWriter(csvfile, output_names, output_types, DELIMITER)
    writer.writeheader()
    
    i = 1
    keep_running = True
    while keep_running:
        
        if i%args.frequency == 0:
            if args.samples > 0:
                sys.stdout.write("\r")
                sys.stdout.write("{:.2%} done.".format(float(i)/float(args.samples))) 
                sys.stdout.flush()
            else:
                sys.stdout.write("\r")
                sys.stdout.write("{:3d} samples.".format(i)) 
                sys.stdout.flush()
        if args.samples > 0 and i >= args.samples:
            keep_running = False
        try:
            state = con.receive()
            if state is not None:
                writer.writerow(state)
            else:
                sys.exit()
        except KeyboardInterrupt:
            keep_running = False
        i += 1

sys.stdout.write("\rComplete!            \n")

con.send_pause()
con.disconnect()

You need to modify the following parameters in your own code:
parser.add_argument('--host', default='192.168.1.1', help='name of host to connect to (localhost)')
parser.add_argument('--config', default='pose_vel_logger_config.xml', help='data configuration file to use (record_configuration.xml)')
parser.add_argument('--output', default='robot_data.csv', help='data output file to write to (robot_data.csv)')

stevensu1838, 2019-05-28:
Hi all,
For any of you who wants to read data in real-time from the TCP of UR5, please check out the following site:
https://www.universal-robots.com/how-tos-and-faqs/how-to/ur-how-tos/real-time-data-exchange-rtde-guide-22229/
To read the data, please run the record.py code as shown below:
import argparse
import logging
import sys
sys.path.append('..')
import rtde.rtde as rtde
import rtde.rtde_config as rtde_config
import rtde.csv_writer as csv_writer
DELIMITER = ","
#parameters
parser = argparse.ArgumentParser()
parser.add_argument('--host', default='192.168.1.1', help='name of host to connect to (localhost)')
parser.add_argument('--port', type=int, default=30004, help='port number (30004)')
parser.add_argument('--samples', type=int, default=0, help='number of samples to record')
parser.add_argument('--frequency', type=int, default=125, help='the sampling frequency in Herz')
parser.add_argument('--config', default='pose_vel_logger_config.xml', help='data configuration file to use (record_configuration.xml)')
parser.add_argument('--output', default='robot_data.csv', help='data output file to write to (robot_data.csv)')
parser.add_argument("--verbose", help="increase output verbosity", action="store_true")
args = parser.parse_args()
if args.verbose:
logging.basicConfig(level=logging.INFO)
conf = rtde_config.ConfigFile(args.config)
output_names, output_types = conf.get_recipe('out')
con = rtde.RTDE(args.host, args.port)
con.connect()
get controller version
con.get_controller_version()
setup recipes
if not con.send_output_setup(output_names, output_types, frequency = args.frequency):
logging.error('Unable to configure output')
sys.exit()
#start data synchronization
if not con.send_start():
logging.error('Unable to start synchronization')
sys.exit()
with open(args.output, 'w') as csvfile:
writer = csv_writer.CSVWriter(csvfile, output_names, output_types, DELIMITER)
writer.writeheader()
i = 1
keep_running = True
while keep_running:
    
    if i%args.frequency == 0:
        if args.samples > 0:
            sys.stdout.write("\r")
            sys.stdout.write("{:.2%} done.".format(float(i)/float(args.samples))) 
            sys.stdout.flush()
        else:
            sys.stdout.write("\r")
            sys.stdout.write("{:3d} samples.".format(i)) 
            sys.stdout.flush()
    if args.samples > 0 and i >= args.samples:
        keep_running = False
    try:
        state = con.receive()
        if state is not None:
            writer.writerow(state)
        else:
            sys.exit()
    except KeyboardInterrupt:
        keep_running = False
    i += 1
sys.stdout.write("\rComplete! \n")
con.send_pause()
con.disconnect()
You need to modify the following parameters in your own code:
parser.add_argument('--host', default='192.168.1.1', help='name of host to connect to (localhost)')
parser.add_argument('--config', default='pose_vel_logger_config.xml', help='data configuration file to use (record_configuration.xml)')
parser.add_argument('--output', default='robot_data.csv', help='data output file to write to (robot_data.csv)')

Btw, I'd love to say thank you to Andy for his great work. Cheers

AndyZe, 2019-05-28:
Thanks Steven! This might come in handy

closed 2019-05-29 by stevensu1838