UTNuclearRoboticsPublic/jog_arm #101: What is the current state of this project

farmange, 2019-07-09:
Hi,
I am really interesting in the Jog feature and I am currently trying the code on a simulated UR5 and ROS Kinetic. 
In the meantime, I realize that you are working on jog package integration into moveit_experimental. It's a great thing and I would like to test this last version! 
However, there is nothing about jog_arm on kinetic or kinetic-devel branch of the [moveit](https://github.com/ros-planning/moveit/tree/kinetic-devel/moveit_experimental) repo. 
How can I proceed to test on Kinetic ? 
Thank you for your help

AndyZe, 2019-07-09:
This repo should still work on Kinetic. Or you can grab the version from moveit_experimental master branch -- it still runs on Kinetic too. The version from moveit_experimental has some improvements that aren't here.

To get jogging without a bunch of other MoveIt junk, just clone or download the whole moveit repo then copy/paste moveit_experimental into your catkin workspace. It's too bad it's not released as a debian anymore.

AndyZe, 2019-07-09:
This pull request has specific instructions to run on a simulated UR5:  https://github.com/ros-planning/moveit/pull/1482

closed 2019-07-10 by AndyZe

farmange, 2019-07-10:
Hi @AndyZe, 

Indeed, this repo works on Kinetic but I would like the new things and the refactoring you did on moveit_experimental. 

I tried to directly copy the moveit_experimental (from your branch AndyZe:use_ur3_example) on my catkin workspace but there is a circular dependency 
` 
CMake Error at /home/ros/workspace/catkin_ws_jog_andyze/build/catkin_generated/order_packages.cmake:7 (message):
  Circular dependency in subset of packages:

  chomp_motion_planner, moveit_experimental, moveit_kinematics,
  moveit_ros_manipulation, moveit_ros_move_group, moveit_ros_planning,
  moveit_ros_planning_interface, moveit_ros_warehouse
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_workspace.cmake:42 (include)
  CMakeLists.txt:67 (catkin_workspace)
`
When I directly copy the content of the moveit_experimental folder inside my catkin workspace it builds. To do that I have to replace the original CMakeLists.txt of my catkin workspace by the one provide in moveit_experimental folder. I don't know the impact of that or if it could break something. I am not sure it is the way to do. I am not familiar with ROS structure yet...

Anyway, I followed the PR README.md for the simulated UR5.
I get some warnings when I ran `ur5_moveit_planning_execution.launch` and `moveit_rviz.launch` :
`
[` WARN] ros.moveit_core.robot_model: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
`
Then when I listed controllers there was no joint_group_position_controller on my robot. Only joint_state_controller and arm_controller. 

I modified the ur_simulated_config.yaml file with the following lines :
`
command_out_type: trajectory_msgs/JointTrajectory
command_out_topic: arm_controller/command
`
I also removed the spacenav_node node as I only have xbox controller with me...

I was not really confident but the arm finally moves ! 

For now, I am not familiar with controller (if you have good course I would be happy) and I would like to understand what is inside joint_group_position_controller. My open question :
Why this controller is missing on my UR ? 
Why there are warnings about base_link and world frame ?

Sorry for the long story and I hope to go further with your code !

AndyZe, 2019-07-11:
Farmange,

Glad you got it working! I don't know why the joint_group_position_controller is missing for you. However, it was added in this PR:

https://github.com/ros-industrial/universal_robot/pull/422

So if you update your universal_robot repo (kinetic-devel branch) it should be present.

farmange, 2019-07-12:
Yes, you right. 
I managed to get joint_group_position_controller on universal_robot repo kinetic-devel branch.
Thank you again