UTNuclearRoboticsPublic/jog_arm #105: Can not jog with real UR3

Haoran-Zhao, 2020-04-15:
Hi, I tried to jogging with the real UR3 and Xbox controller. it can be done in the simulation. However, there was no response when I implemented it on a real UR3.  I have used the 
[ur3_jog.yaml.txt](https://github.com/UTNuclearRoboticsPublic/jog_arm/files/4482316/ur3_jog.yaml.txt) file you provided and did the same thing as following:

> Damn right, now it's works!
> 
> 1. In [ur3_ros_control](https://github.com/ros-industrial/ur_modern_driver/blob/master/launch/ur3_ros_control.launch) need need to stop all controllers, except joint_group_vel_controller. Of course you can do this with `rosrun rqt_controller_manager rqt_controller_manager`.
> 2. In my case:
> 
> ```
> command_out_topic: joint_group_vel_controller/command
> command_out_type: std_msgs/Float64MultiArray
> ```
> 
> and publish only joint velocity.
> 3. Launch this (obviously with yours right robot_ip):
> 
> ```
> <?xml version="1.0"?>
> <launch>
>   <include file="$(find ur_modern_driver)/launch/ur3_ros_control.launch">
>     <arg name="robot_ip" value="192.168.102.100"/>
>   </include>
> 
>   
>   <include file="$(find ur3_moveit_config)/launch/move_group.launch">
>     <arg name="limited" value="true"/>
>   </include>
> 
>   <include file="$(find ur3_moveit_config)/launch/moveit_rviz.launch">
>     <arg name="config" value="true"/>
>   </include>
> 
> </launch>
> ```
> 
> And you `jog_with_something` launch-file.
> 
> P.S. Sometimes Rviz falls, but this is a completely different question. Thanks again for the right direction @AndyZe )

[image]
[image]
There is no connection between `joint_group_vel_controller` and 'move_group'. I don't know if it's right?

However, the UR3 did not move when I give the command with Xbox controller, and it shows:
```
[ WARN] [1586885701.872903117]: Stale or zero command. Try a larger 'incoming_command_timeout' parameter?
[ WARN] [1586885703.880898942]: Stale or zero command. Try a larger 'incoming_command_timeout' parameter?
[ WARN] [1586885705.888888321]: Stale or zero command. Try a larger 'incoming_command_timeout' parameter?
[ WARN] [1586885707.896932954]: Stale or zero command. Try a larger 'incoming_command_timeout' parameter?
```
I changed "incoming_command_timeout" parameter from 0.2 to 10, it always gave me this error.

I `rostopic echo /joy`, the xbox controller works, the input values can be published to `/jog_arm_server/delta_jog_cmds`, and there are data in `/joint_group_vel_controller/command`. However, UR3 has no response.

would you mind give me some suggestions to solve it? Thanks a lot!

AndyZe, 2020-04-16:
No connection between `joint_group_vel_controller` and `move_group` is OK.

AndyZe, 2020-04-16:
The ur_modern_driver package provides that controller here:

https://github.com/ros-industrial/ur_modern_driver/blob/kinetic-devel/config/ur5_controllers.yaml

Haoran-Zhao, 2020-04-16:
@AndyZe Thanks for your apply, I have done the `rosservice call controller_manager list_controllers` and the `joint_group_vel_controller` was there. I almost checked all your response related to this topic. However, it still can not drive the ur3.
Maybe it's the version of ubuntu or the ROS-kinetic? I cloned the repository from moveit, but I can not catkin_make it. I think one problem is the Cmake request version is 3.1.3 rather than 2.8.3 on my laptop. I fixed all of them, it still reports an error. 
I found a solution to publish the desired joint velocity to the `/joint_speed`, which solves the problem. But I don't know if this is the right way or it's another approach.