UTNuclearRoboticsPublic/jog_arm #106: Question on Axis Mapping for SpaceMouse

luke-kim409, 2020-04-25:
Hello, @AndyZe 
Thank you for your great work!

I initially found the more recent version of this package, [moveit_jog_arm](https://ros-planning.github.io/moveit_tutorials/doc/arm_jogging/arm_jogging_tutorial.html), but I found that the moveit_jog_arm package works only with ROS Melodic, but not Kinetic (**could you please confirm whether the new version works on Kinetic as well as Melodic?**). This older version works perfectly well with ROS Kinetic, which I have to use instead of Melodic for now to control my KUKA LBR iiwa robot based on [iiwa_stack](https://github.com/IFL-CAMP/iiwa_stack) due to some connection problem on Melodic.

I am wondering what would be the best way to change or customize the mapping of axes of 3D mouse for my robot configuration. For moveit_jog_arm package that works on Melodic, the **[spacenav_via_teleop_tools.yam](https://github.com/ros-planning/moveit/blob/master/moveit_experimental/moveit_jog_arm/config/spacenav_via_teleop_tools.yaml)**l file allows one to customize the axes mapping, and it worked perfectly well when I tried it without my physical robot (which unfortunately cannot be connected in ROS Melodic for now). 

For this older jog_arm package, which uses [spacenav_to_twist.cpp](https://github.com/UTNuclearRoboticsPublic/jog_arm/blob/kinetic/jog_arm/src/jog_arm/teleop_examples/spacenav_to_twist.cpp) for converting the joystick commands to jog commands, I thought that the following block of the code can be modified to change the axes mapping.

void joyCallback(const sensor_msgs::Joy::ConstPtr& msg)
  {
    // Cartesian jogging with the axes
    geometry_msgs::TwistStamped twist;
    twist.header.stamp = ros::Time::now();
    twist.twist.linear.x = msg->axes[0];
    twist.twist.linear.y = msg->axes[1];
    twist.twist.linear.z = msg->axes[2];

    twist.twist.angular.x = msg->axes[3];
    twist.twist.angular.y = msg->axes[4];
    twist.twist.angular.z = msg->axes[5];

    // Joint jogging with the buttons
    jog_msgs::JogJoint joint_deltas;
    // This example is for a Motoman SIA5. joint_s is the base joint.
    joint_deltas.joint_names.push_back("joint_s");

    // Button 0: positive on the wrist joint
    // Button 1: negative on the wrist joint
    joint_deltas.deltas.push_back(msg->buttons[0] - msg->buttons[1]);
    joint_deltas.header.stamp = ros::Time::now();

    twist_pub_.publish(twist);
    joint_delta_pub_.publish(joint_deltas);
  }

For example, like below:

    twist.twist.linear.x = - msg->axes[1];
    twist.twist.linear.y = msg->axes[0];
    twist.twist.linear.z = msg->axes[2];

    twist.twist.angular.x = - msg->axes[4];
    twist.twist.angular.y = msg->axes[3];
    twist.twist.angular.z = msg->axes[5];

This, however, didn't make any change.

Also, even if I change the specified _joint name_ for "joint jogging with buttons", for example,  
from **joint_deltas.joint_names.push_back("joint_s");** to **joint_deltas.joint_names.push_back("joint_7");**, nothing changes. 

rostopic echo /jog_arm_server/joint_delta_jog_cmds still shows "joint_s" instead of "joint_7" as follows:

header: 
  seq: 7667
  stamp: 
    secs: 1587848605
    nsecs: 251340068
  frame_id: ''
joint_names: [joint_s]
deltas: [0.0]
---

Could you please help me figure out how to remap the joystick axes and specify the joint to be jogged with the buttons? 

Many thanks again for your great work!

AndyZe, 2020-04-25:
I think you're on the right track! Maybe you forgot to run catkin_make again, or open a new terminal, or your ROS workspace isn't sourced correctly

luke-kim409, 2020-04-25:
Thank you so much for the very quick and accurate answer, @AndyZe!
You are absolutely right! After I ran catkin build on my workspace, the axes mapping started working as intended! Thanks a lot!

closed 2020-04-25 by luke-kim409